JeffyCN/meta-rockchip #104: question: chromium v120 decode 4K

lilulu2023, 2024-05-10:
Dear Jeffy , we have successfully built chromium with your patch. videos in chromium with 2K resolution play well but 4K still not satisfying, we noticed that with kylin attached chromium version 83 works fine with the same 4K video. So what should we do to figure out whats different between 83 and our own 120 build ? Besides, both version recognized GPU mali T860. Thank you for any help.

JeffyCN, 2024-05-10:
the r119+ requires newer libv4l-rkmpp

lilulu2023, 2024-05-11:
@JeffyCN I cloned mpp  and  libv4l-rkmpp  library and built them copied libv4l-rkmpp.so to /usr/lib/libv4l/plugins/, after that I  run custom built chromium with ```export XDG_RUNTIME_DIR=/run/user/0``` and needed flags but seemed still not working. How can I figure out which part is not working?

JeffyCN, 2024-05-11:
please check the plugin's readme for details

lilulu2023, 2024-05-13:
@JeffyCN I set noticed  LIBV4L_RKMPP_LOG_LEVE0x10 but found that ```libv4l-rkmpp version``` is never logged. 
Seems chromium is not calling it ? 
Besides , setting export XDG_RUNTIME_DIR=/run/user/0 will result permission problems and if I run with root, will otherwise call libGL.so permission problems.  
One more thing, /usr/lib/libv4l/plugins/ does not exist, I created libv4l/plugins and copied libv4l-rkmpp.so into it,  is this ok ?

lilulu2023, 2024-05-14:
I cloned  v4l-utils and used branch stable-1.20 to apply the 0001-0003 patches for v4l-utils in master branch of this repo. I then installed v4l according to its instructions. Then I rebooted device and start chrome again. But seems I still have no /usr/lib/v4l/plugins directory created. Do the versions matter ? If not, anything I can do to figure this out ?

JeffyCN, 2024-05-14:
that depends on your target system. if it supports multi-arch, the dir could be /usr/lib/*/v4l*/plugins/

lilulu2023, 2024-05-14:
Actually not using the sdk， I am doing it directly in the onboard system, which is debian11(bullseye)

JeffyCN, 2024-05-14:
debian uses multiarch dirs.

you can use 'dpkg-buildpackage -b -us -uc' to build libv4l & v4l-mpp deb

lilulu2023, 2024-05-14:
I am doing the build inside a arm64 debian on rockchip SBC, I noticed that a log "Failed to query video capabilities" when running custom built chomium. That log I think should be from libv4l-utils/lib/libv4l because I found it in the source code. But after I add another log of my own and run make && sudo make install inside libv4l-utils, then run custom built chomium , there is no extra log of my own shown, which means my built of libv4l-utils is not installed or called. I also noticed that there is a dso libv4l.so in /usr/lib but build outputs of libv4l-utils contain nothing like this. so I guess the chromium is using  liv4l-util via libv4l.so ? Or can you please publish liv4l-utils version you used for testing ?

JeffyCN, 2024-05-14:
all version are fine.

if you are using rockchip debian sdk, please report issue to rockchip redmine, the custom libv4l should be installed by default.

JeffyCN, 2024-05-14:
on rockchip debian sdk, the related parts:
```
/etc/init.d/rockchip.sh:
# Create dummy video node for chromium V4L2 VDA/VEA with rkmpp plugin
echo dec > /dev/video-dec0
echo enc > /dev/video-enc0
chmod 660 /dev/video-*
chown root:video /dev/video-*

# The chromium using fixed pathes for libv4l2.so
ln -rsf /usr/lib/*/libv4l2.so /usr/lib/
[ -e /usr/lib/aarch64-linux-gnu/ ] && ln -Tsf lib /usr/lib64
```

JeffyCN, 2024-05-14:
[v4l.tar.gz](https://github.com/JeffyCN/meta-rockchip/files/15305624/v4l.tar.gz)
prebuilt v4l packages attached.

lilulu2023, 2024-05-21:
Closing this for we figured out using an older version.

closed 2024-05-21 by lilulu2023