JeffyCN/meta-rockchip #107: OpenGL issue in glXChooseVisual

berkutta, 2024-05-15:
I'm running a custom distro with this meta layer on Rockchip PX30 Hardware.

Choosing OpenGL visuals seem to be not working. I'm not sure if this is a meta-rockchip issue or just some missing configuration/package.

Code like this returns NULL and trigger the Assert:
```
GLint attr[] = {GLX_RGBA, GLX_DEPTH_SIZE, 24, GLX_DOUBLEBUFFER, None};
XVisualInfo* visual = glXChooseVisual(pLayer->display, screen, attr);
ASSERT(visual != NULL);
```

Also tools like glxinfo and glxdemo are not working. Is this a known issue?

```
root@px30:/var/lib/videoengine# DISPLAY=:0 glxinfo
name of display: :0
Error: couldn't find RGB GLX visual or fbconfig

root@px30:/var/lib/videoengine# DISPLAY=:0 glxdemo
Error: couldn't get an RGB, Double-buffered visual
```

JeffyCN, 2024-05-15:
the mali gpu ddk doesn't support glx.

it supports opengles2

berkutta, 2024-05-16:
Ok, thanks

closed 2024-05-16 by berkutta