JeffyCN/meta-rockchip #108: 6.1内核编译报错

sensiki, 2024-05-16:
ERROR: Logfile of failure stored in: /home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/temp/log.do_compile.2061871
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: KBUILD_BUILD_TIMESTAMP: Fri Dec 29 10:03:45 UTC 2023
| NOTE: make -j 48 CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source=/usr/src/debug/linux-rockchip/6.1  -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source=/usr/src/debug/linux-rockchip/6.1  -fmacro-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/linux-rockchip_rk3568_evb-standard-build=/usr/src/debug/linux-rockchip/6.1  -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/linux-rockchip_rk3568_evb-standard-build=/usr/src/debug/linux-rockchip/6.1  -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot=  -fmacro-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot=  -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native=  -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-build-artifacts=/usr/src/kernel  --sysroot=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot LD=aarch64-poky-linux-ld.bfd    --sysroot=/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot OBJCOPY=aarch64-poky-linux-objcopy  STRIP=aarch64-poky-linux-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/usr/lib                         -L/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/lib                         -Wl,-rpath,/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/sparks/yocto-scarthgap/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/home/sparks/yocto-scarthgap/build/tmp/work/rockchip_rk3568_evb-poky-linux/linux-rockchip/6.1/recipe-sysroot-native/usr/include -O2 -pipe HOSTPKG_CONFIG=pkg-config-native PAHOLE=false -j 48 rk3568-evb1-ddr4-v10-linux.img
|   GEN     Makefile
|   CALL    /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/checksyscalls.sh
|   CC      drivers/media/platform/rockchip/cif/capture.o
| fatal: not a git repository (or any of the parent directories): .git
| /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/drivers/media/platform/rockchip/cif/capture.c:4929:6: error: conflicting types for 'rkcif_do_stop_stream' due to enum/integer mismatch; have 'void(struct rkcif_stream *, unsigned int)' [-Werror=enum-int-mismatch]
|  4929 | void rkcif_do_stop_stream(struct rkcif_stream *stream,
|       |      ^~~~~~~~~~~~~~~~~~~~
| In file included from /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/drivers/media/platform/rockchip/cif/capture.c:24:
| /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/drivers/media/platform/rockchip/cif/dev.h:934:6: note: previous declaration of 'rkcif_do_stop_stream' with type 'void(struct rkcif_stream *, enum rkcif_stream_mode)'
|   934 | void rkcif_do_stop_stream(struct rkcif_stream *stream,
|       |      ^~~~~~~~~~~~~~~~~~~~
| /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/drivers/media/platform/rockchip/cif/capture.c:6090:5: error: conflicting types for 'rkcif_do_start_stream' due to enum/integer mismatch; have 'int(struct rkcif_stream *, unsigned int)' [-Werror=enum-int-mismatch]
|  6090 | int rkcif_do_start_stream(struct rkcif_stream *stream, unsigned int mode)
|       |     ^~~~~~~~~~~~~~~~~~~~~
| /home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/drivers/media/platform/rockchip/cif/dev.h:932:5: note: previous declaration of 'rkcif_do_start_stream' with type 'int(struct rkcif_stream *, enum rkcif_stream_mode)'
|   932 | int rkcif_do_start_stream(struct rkcif_stream *stream,
|       |     ^~~~~~~~~~~~~~~~~~~~~
| cc1: all warnings being treated as errors
| make[7]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:250: drivers/media/platform/rockchip/cif/capture.o] Error 1
| make[6]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:500: drivers/media/platform/rockchip/cif] Error 2
| make[5]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:500: drivers/media/platform/rockchip] Error 2
| make[4]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:500: drivers/media/platform] Error 2
| make[3]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:500: drivers/media] Error 2
| make[2]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/scripts/Makefile.build:500: drivers] Error 2
| make[1]: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/Makefile:2012: .] Error 2
| make: *** [/home/sparks/yocto-scarthgap/build/tmp/work-shared/rockchip-rk3568-evb/kernel-source/arch/arm64/Makefile:223: rk3568-evb1-ddr4-v10-linux.img] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/sparks/yocto-scarthgap/build/../meta-rockchip/recipes-kernel/linux/linux-rockchip_6.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3087 tasks of which 2869 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/sparks/yocto-scarthgap/build/../meta-rockchip/recipes-kernel/linux/linux-rockchip_6.1.bb:do_compile
Summary: There were 2 ERROR messages, returning a non-zero exit code.

JeffyCN, 2024-05-16:
sorry, the kernel repo is just a mirror repo, and i'm not allowed to provide newer sources.

maybe you can try to find newer kernel in other repo(for example radxa) or use rockchip's official sdk. you can follow the readme here to use a local custom kernel source.

for this error only, maybe try to modify the code(drivers/media/platform/rockchip/cif/capture.c) and change 'unsigned int' to 'emum rkcif_stream_mode'

closed 2024-05-16 by JeffyCN

sensiki, 2024-05-16:
Thank you for your answer

JeffyCN, 2024-05-16:
and maybe you can try kernel-5.10 too.

for this issue, i do find a patch, i've uploaded it just now.