JeffyCN/meta-rockchip #109: Building scarthgap with INIT_MANAGER = "systemd" fails in QA because of missing upstream status

asac, 2024-05-27:
ERROR: systemd-1_255.4-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/panta/yocto/panta/meta/meta-pantavisor/build/../meta-rockchip/recipes-core/systemd/systemd/0001-meson-do-not-fail-build-with-newer-kernel-headers.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: systemd-1_255.4-r0 do_patch: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/panta/yocto/panta/meta/meta-pantavisor/build/tmp-scarthgap/work/armv8a-poky-linux/systemd/255.4/temp/log.do_patch.3220363
ERROR: Task (/home/panta/yocto/panta/meta/meta-pantavisor/build/../poky/meta/recipes-core/systemd/systemd_255.4.bb:do_patch) failed with exit code '1'

JeffyCN, 2024-05-27:
uploaded a fix just now :)

asac, 2024-05-27:
@JeffyCN thanks for the quick turnaround. the meson patch for systemd has the same issue popping up now that the other is fixed.

asac, 2024-05-27:
sorry my comment before was due to a glich in the matrix. this is fixed. thanks

closed 2024-05-27 by asac