JeffyCN/meta-rockchip #110: no-prefix rootfs.img link breaks initramfs builds

asac, 2024-05-27:
When building initramfs the machine conf level POST_PROCESS command:

IMAGE_POSTPROCESS_COMMAND:append = " link_rootfs_image;"
link_rootfs_image() {
        ln -sf "${IMAGE_LINK_NAME}.${RK_ROOTFS_TYPE}" \
                "${IMGDEPLOYDIR}/rootfs.img"
}

Creates a conflict in deploy dir with the actual images (e.g. core-image-minimal).

Is that link really necessary? Who is the consumer of that rootfs.img file? (I couldnt find one in meta-*). Can we do without this rootfs.img?

JeffyCN, 2024-05-27:
well, that is for generating rockchip specified update image. you can remove it(and some other later functions as well)

asac, 2024-05-27:
OK, I think https://github.com/JeffyCN/meta-rockchip/commit/a74e2fa9934279d27fc6b223cae8ac43fa63ef6f fixes this one as well. Thanks!

closed 2024-05-27 by asac