JeffyCN/meta-rockchip #111: IMAGE_FSTYPES:append is forcing wic onto initramfs

asac, 2024-05-27:
When using IMAGE_FSTYPES:append instead of += we observe that the usual pattern to set:

IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"

In the initramfs image recipe does not properly remove wic image etc....

A patch I did locally is like:

From a7b3601df12caee21ccd3aac3d5753695493444e Mon Sep 17 00:00:00 2001
From: Alexander Sack <asac@pantacor.com>
Date: Mon, 27 May 2024 08:30:19 +0000
Subject: [PATCH] fix IMAGE_FSTYPES

---
 conf/machine/include/rockchip-common.inc | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/conf/machine/include/rockchip-common.inc b/conf/machine/include/rockchip-common.inc
index 578e2ba..363b269 100644
--- a/conf/machine/include/rockchip-common.inc
+++ b/conf/machine/include/rockchip-common.inc
@@ -61,7 +61,7 @@ RDEPENDS:${KERNEL_PACKAGE_NAME}-base = ""
 IMAGE_FSTYPES:remove = "iso live"
 export RK_ROOTFS_TYPE ?= "ext4"

-IMAGE_FSTYPES:append = " ${RK_ROOTFS_TYPE} wic"
+IMAGE_FSTYPES ?= "${RK_ROOTFS_TYPE} wic"
 WKS_FILE ?= "generic-gptdisk.wks.in"

 # Some partitons, e.g. trust, are allowed to be optional.
-- 
2.34.1

Alternatively, using += is also more humble and allows the downstream recipe to unset the IMAGE_FSTYPES variable properly.

From a7b3601df12caee21ccd3aac3d5753695493444e Mon Sep 17 00:00:00 2001
From: Alexander Sack <asac@pantacor.com>
Date: Mon, 27 May 2024 08:30:19 +0000
Subject: [PATCH] fix IMAGE_FSTYPES

---
 conf/machine/include/rockchip-common.inc | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/conf/machine/include/rockchip-common.inc b/conf/machine/include/rockchip-common.inc
index 578e2ba..363b269 100644
--- a/conf/machine/include/rockchip-common.inc
+++ b/conf/machine/include/rockchip-common.inc
@@ -61,7 +61,7 @@ RDEPENDS:${KERNEL_PACKAGE_NAME}-base = ""
 IMAGE_FSTYPES:remove = "iso live"
 export RK_ROOTFS_TYPE ?= "ext4"

-IMAGE_FSTYPES:append = " ${RK_ROOTFS_TYPE} wic"
+IMAGE_FSTYPES += "${RK_ROOTFS_TYPE} wic"
 WKS_FILE ?= "generic-gptdisk.wks.in"

 # Some partitons, e.g. trust, are allowed to be optional.
-- 
2.34.1

JeffyCN, 2024-05-27:
i'm preparing to make them optional, just made patch, verifing it.

JeffyCN, 2024-05-27:
FYI: patch uploaded

asac, 2024-05-27:
Thanks! Looks good now.

closed 2024-05-27 by asac