JeffyCN/meta-rockchip #113: Glibc:do_configure fails due to old kernel headers

AlphaGX, 2024-06-05:
From a fresh install (cloned poky, meta-oe, meta-rockchip) glibc.bb do_configure step fails.

Reproduce setup:
`
mkdir yocto
cd yocto
git clone git://git.yoctoproject.org/poky -b scarthgap
git clone git://git.openembedded.org/meta-openembedded.git -b scarthgap
git clone -b scarthgap https://github.com/JeffyCN/meta-rockchip

source poky/oe-init-build-env
echo 'INHERIT += "rm_work"' >> conf/local.conf
bitbake-layers add-layer ../meta-openembedded/meta-oe/
bitbake-layers add-layer ../meta-rockchip/
MACHINE="rockchip-rk3328-evb" bitbake glibc
`

Revelant part of the error message:
`
| checking installed Linux kernel header files... 3.2.0 or later
| checking for kernel header at least 5.15... too old!
| configure: error: *** The available kernel headers are older than the requested
| *** compatible kernel version
`
Full log of command `MACHINE="rockchip-rk3328-evb" bitbake glibc 2>&1 > glibc.log`: https://pastebin.com/NyVxfXqi

JeffyCN, 2024-06-05:
i thought it should be fixed in https://github.com/JeffyCN/meta-rockchip/commit/51d2ccfb9d5aeb04a296982702e277e4209e0d0f

AlphaGX, 2024-06-07:
I don't know  what to say really, I am using the latest commits so It shouldn't be an issue...
Should I roll back to the commit you mentioned ??

JeffyCN, 2024-06-07:
Or modify there:
pokymeta/conf/bitbake.conf:470:OLDEST_KERNEL = "5.15

JeffyCN, 2024-06-13:
closing this since i cannot repro it.

closed 2024-06-13 by JeffyCN