JeffyCN/meta-rockchip #115: rk3588, chromium 122 HW acceleration and Kernel 6.1

chrisiberg, 2024-06-21:
Should this work since [v4l-rkmpp: Bump to 1.7.1]
,did someone try it?

chrisiberg, 2024-06-24:
Scarthgap needs at least kernel 5.15 for glibc_2.39, so I can't try it with the 5.10 and with 6.1 Kernel Video HW-acceleration does not work.
@JeffyCN Do you see any reason why this would not work with Scarthgap and Kernel 6.1?

JeffyCN, 2024-06-24:
it doesn't require 5.15, it's configuable.

it works well on my board with my buildroot sdk + yocto chromium.

but i am not allowed go provide newer kernel code and some other codes, so i'm not sure would other parts affect it.

JeffyCN, 2024-06-24:
maybe you can provide your rootfs image, i might try to test it with my internal 6.1 kernel to figure out which part went wrong. and maybe you can try to find other newer bsp kernel 6.1 code on other repo(for example radxa)

chrisiberg, 2024-06-24:
That would be great!
To be clear, in chrome://gpu everything looks right, but if I inspect the video with STRG+SHIFT+I > More Tools > Media
Decoder name: VpxVideoDecoder or sometimes FFmpeg
Hardware decoder: **flase**
and CPU usage of chrome is much higher than with kirkstone and kernel 5.10

This is the Kernel source I used, because my hardware is a NanoPC-T6
https://github.com/friendlyarm/kernel-rockchip/tree/nanopi6-v6.1.y

JeffyCN, 2024-06-25:
i cannot access that rootfs.img, no permission

chrisiberg, 2024-06-25:
sorry, should work now.

JeffyCN, 2024-06-27:
please try to upgrade the mpp :)

chrisiberg, 2024-06-27:
Thnx a lot, I will try!

the changes from WORKDIR > UNPACKDIR does not work in all cases I had to change it in 
xserver-xorg for modesetting.conf and in
udev-conf-rocchip for 99-rockchip-permissions.rules

maybe you have a look at it.

chrisiberg, 2024-06-27:
mpp upgrade does not make a difference 
[image]

JeffyCN, 2024-06-28:
1/ Not sure about UNPACKDIR error, it's required by the newest poky, and it works well on my side. Please check your poky version

2/ It works with your image on my board after upgraded mpp and libmali.
That is because the meta-rockchip's kernel is old(and i'm not allowed to upgrade it). If your are using a new kernel, you can need to choose a matched mpp and libmali(valhall-g610-g13p0 for newest kernel)

chrisiberg, 2024-06-28:
Thnx a lot Jeffy!
I got it, this is working!

closed 2024-06-28 by chrisiberg