JeffyCN/meta-rockchip #116: scarthgap not building because of missing xserver version

ursfassler, 2024-06-21:
Poky [updated xserver to 21.1.12](https://git.yoctoproject.org/poky/commit/?h=scarthgap&id=d7cbee7a502dec0deea2dfe12806e608d0d4ca56) due to security vulnerabilities. This version is missing on https://github.com/JeffyCN/xorg-xserver.
This means, Yocto can not be built with the latest scarthgap versions.

Would be great when this is fixed.
Maybe a solution is to use patches for xserver within this repo, that may be more stable against version changes in the future.

chrisiberg, 2024-06-21:
temporary you can change this line.
SRC_URI:append = " git://github.com/JeffyCN/xorg-xserver;protocol=https;nobranch=1;branch=21.1.11;"
in
/meta-rockchip/recipes-graphics/xorg-xserver/xserver-xorg_%.bbappend

JeffyCN, 2024-06-24:
the newest is 21.1.13(i've uploaded that)

and there're too many patches, they might conflict with newer versions.

closed 2024-06-24 by JeffyCN

ursfassler, 2024-06-25:
Thanks for the fast fix @JeffyCN ! :slightly_smiling_face: 

> and there're too many patches, they might conflict with newer versions.

I tested to just apply the patches to the new version, it didn't work out of the box.

ursfassler, 2024-06-25:
Yes, it looks good (not yet finished building it).

My comment was about applying the patches to the new version.