JeffyCN/meta-rockchip #118: Build not giving expecting binaries/outputs

HennieLombaard, 2024-07-19:
Hi

I have setup a new yocto scarthgap build, and followed the build instructions. I am building a core-image-minimal and the build succeeds, but at the end of the build there isn't a wic or update.img in the output directory.

My BBLAYERS
`# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBLAYERS ?= " \
  ${BSPDIR}/poky/meta \
  ${BSPDIR}/poky/meta-poky \
  ${BSPDIR}/poky/meta-yocto-bsp \
  ${BSPDIR}/sources-common/meta-openembedded/meta-oe \
  ${BSPDIR}/sources-common/meta-rockchip \
  "`

In local.conf I only changed 
`MACHINE = "rockchip-rk3566-evb"`

Build succeeds without issue, but output directory (../tmp/deploy/images/rockchip-rk3566-evb) doesn't contain the wic or update.img
`tmp/deploy/images/rockchip-rk3566-evb$ ls
boot.img
boot.img--4.19-r0-rockchip-rk3566-evb-20240718133225.bin
boot.img-rockchip-rk3566-evb.bin
core-image-full-cmdline-rockchip-rk3566-evb.rootfs-20240719072145.manifest
core-image-full-cmdline-rockchip-rk3566-evb.rootfs-20240719072145.spdx.tar.zst
core-image-full-cmdline-rockchip-rk3566-evb.rootfs-20240719072145.tar.gz
core-image-full-cmdline-rockchip-rk3566-evb.rootfs-20240719072145.testdata.json
core-image-full-cmdline-rockchip-rk3566-evb.rootfs.manifest
core-image-full-cmdline-rockchip-rk3566-evb.rootfs.spdx.tar.zst
core-image-full-cmdline-rockchip-rk3566-evb.rootfs.tar.gz
core-image-full-cmdline-rockchip-rk3566-evb.rootfs.testdata.json
core-image-minimal-rockchip-rk3566-evb.rootfs-20240718133225.manifest
core-image-minimal-rockchip-rk3566-evb.rootfs-20240718133225.spdx.tar.zst
core-image-minimal-rockchip-rk3566-evb.rootfs-20240718133225.tar.gz
core-image-minimal-rockchip-rk3566-evb.rootfs-20240718133225.testdata.json
core-image-minimal-rockchip-rk3566-evb.rootfs.manifest
core-image-minimal-rockchip-rk3566-evb.rootfs.spdx.tar.zst
core-image-minimal-rockchip-rk3566-evb.rootfs.tar.gz
core-image-minimal-rockchip-rk3566-evb.rootfs.testdata.json
Image
Image--4.19-r0-rockchip-rk3566-evb-20240718133225.bin
Image-rockchip-rk3566-evb.bin
modules--4.19-r0-rockchip-rk3566-evb-20240718133225.tgz
modules-rockchip-rk3566-evb.tgz
rk3566-evb2-lp4x-v10-linux--4.19-r0-rockchip-rk3566-evb-20240718133225.dtb
rk3566-evb2-lp4x-v10-linux.dtb
rk3566-evb2-lp4x-v10-linux-rockchip-rk3566-evb.dtb
zboot.img
zboot.img--4.19-r0-rockchip-rk3566-evb-20240718133225.bin
zboot.img-rockchip-rk3566-evb.bin`

Any ideas what I am missing here?

JeffyCN, 2024-07-19:
you may need to set this in your local config
https://github.com/JeffyCN/yocto-conf/blob/master/include/common.conf#L293

HennieLombaard, 2024-07-19:
Excellent

Thank you, added that line to my local.conf now its creating the expected files. Thank you. Maybe just add it to the main README, although if I'm the only one its happened to I must have done something wrong somewhere.

Added to local.conf
`INHERIT:append = " rockchip-image"`

Issue resolved, thank you

JeffyCN, 2024-07-19:
done.
that was enabled by default, but changed recently, i forgot to update the readme.

closed 2024-07-19 by JeffyCN