JeffyCN/meta-rockchip #119: EGL issue with weston on scarthgap branch

dylad, 2024-07-31:
Hello,

I am having some issues with EGL.
When building core-image-weston with meta-rockchip (scarthgap) for RK3588S board, examples app from weston using EGL crash with a segfault.  (for instance weston-simple-egl)

I am using lastest revision from scarthgap with poky and meta-openembedded. No others layers are involved.
I just add `DEPENDS:append:pn-xwayland = " libtirpc"` in my local.conf because xwayland seems to have a missing dependency.

I am building an image for RK3588S chip.

JeffyCN, 2024-07-31:
maybe try to apply this:
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/wayland/0003-client-Assign-proxy-to-default-queue-when-releasing-.patch

dylad, 2024-07-31:
I didn't try the manifest but the first patch seems to work on my side !
Thanks @JeffyCN !

dylad, 2024-08-01:
Hi again @JeffyCN 
While the segfault no longer happen, I just notice that weston-simple-egl produces two windows (see following screenshot):

[image]

Then if I hit a Ctrl-C with a keyboard on Weston (not in the shell which started the app obviously), the second window is killed.

[image]

JeffyCN, 2024-08-01:
it works well on my side, maybe try to upload your rootfs.img

dylad, 2024-08-01:
I've started a fresh build, using the manifest this time. I'll see if there is a difference.
Are you also using a Rk3588S-based board ?

dylad, 2024-08-01:
@JeffyCN 
In the mean time, you can find my generated rootfs [here](https://we.tl/t-q8Zl0c3y1H)

JeffyCN, 2024-08-01:
didn't repro it with your rootfs, everything works well on my board...

dylad, 2024-08-01:
It seems I mismatch my rootfs when flashing. Sorry about that.
I'll try to trigger this in a reproducible way on my side and come back to you.
Thanks for your help !

closed 2024-08-01 by dylad

dylad, 2024-08-01:
BTW, I just flashed a fresh image using scarthgap manifest file without the patch you mentioned from Buildroot.
weston-simple-egl segfaults again so I guess this patch is needed using Yocto build.