JeffyCN/meta-rockchip #120: RK3588S with kernel 6.1 egl segfault

BuzzBumbleBee, 2024-08-07:
Hello,

I'm trying to get acceleration working on a radxa CM5 based device (RK3588S), however the vendor only supplies dts for 6.1 kernel used in meta-rockchip.

I'm seeing segfaults on egl tests and Chromium is very unstable.

Should the 6.1 kernel used here work with the same libmali that works with 5.10 ?

Thanks In Advance

BuzzBumbleBee, 2024-08-07:
OK so updating to the g13p0 libmali fixed the e

gl test crashes, but still no chromium video decoding

I have attached the logs, the versions are the same as https://github.com/JeffyCN/yocto-manifests/blob/scarthgap/yocto.xml

[libv4l2_rkmpp_log.txt](https://github.com/user-attachments/files/16534362/libv4l2_rkmpp_log.txt)
[cr_logs.txt](https://github.com/user-attachments/files/16534363/cr_logs.txt)

JeffyCN, 2024-08-07:
the cr log shows gpu process crashed

JeffyCN, 2024-08-07:
libmali doesn't care about kernel version, but should match the mali driver version, i don't know much about that part, but maybe try to use the ones with closest commit date(in mali driver and libmali)

BuzzBumbleBee, 2024-08-08:
@JeffyCN I rebuilt on latest scarthgap branches for poky / oe / clang and master for browser.

This resulted in a fully working build with GPU and Video acceleration on chromium 126.

However chromium will only work with the '--no-sandbox' flag, without it I get permission denied on libEGL.so and linGLESv2.so (both under /use/lib/chromium)

I'm unsure why this is as the filesystem permissions are all world readable and listable for directories.

JeffyCN, 2024-08-09:
the chromium's sandbox has it's own permission system(ioctl blacklist and whitelist)

JeffyCN, 2024-08-09:
it's a little hard to add all required device node into those lists, so i just disable the sandbox instead.

closed 2024-08-12 by JeffyCN