JeffyCN/meta-rockchip #121: is it support RK3588S?

masoudjalalir, 2024-08-12:
hello friends,
i have an orangepi5B, which has RK3588s
is it possible to make image for it(using Yocto project and meta-openembedded?

JeffyCN, 2024-08-12:
this is only for evb boards. for other boards please check the readme about using custom kernel and loader

masoudjalalir, 2024-08-21:
Reading this part " V. Supporting new Machine "?

masoudjalalir, 2024-08-21:
i got these errors

ERROR: linux-rockchip-5.10-r0 do_compile: oe_runmake failed
ERROR: linux-rockchip-5.10-r0 do_compile: ExecutionError('/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/temp/run.do_compile.1522209', 1, None, None)
ERROR: Logfile of failure stored in: /home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/temp/log.do_compile.1522209

Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: KBUILD_BUILD_TIMESTAMP: Sat Jan  8 10:24:03 UTC 2022
| NOTE: make -j 12 CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fcanon-prefix-map  -fmacro-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source=/usr/src/debug/linux-rockchip/5.10  -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source=/usr/src/debug/linux-rockchip/5.10  -fmacro-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/linux-rk3588s-standard-build=/usr/src/debug/linux-rockchip/5.10  -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/linux-rk3588s-standard-build=/usr/src/debug/linux-rockchip/5.10  -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot=  -fmacro-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot=  -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native=  -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source=/usr/src/kernel -fdebug-prefix-map=/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-build-artifacts=/usr/src/kernel  --sysroot=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot LD=aarch64-poky-linux-ld.bfd    --sysroot=/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot OBJCOPY=aarch64-poky-linux-objcopy  STRIP=aarch64-poky-linux-strip HOSTCC=gcc  HOSTCFLAGS=-isystem/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/lib                         -L/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/lib                         -Wl,-rpath,/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/im/Desktop/yocto_BSB/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/include -O2 -pipe HOSTPKG_CONFIG=pkg-config-native PAHOLE=false -j 12 rk3588s-orangepi-5.img
| make[3]: *** No rule to make target 'arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5.dtb'.  Stop.
| make[2]: *** [/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/Makefile:1444: rockchip/rk3588s-orangepi-5.dtb] Error 2
| make[2]: *** Waiting for unfinished jobs....
| /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c: In function 'display_openssl_errors':
| /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c:46:2: warning: 'ERR_get_error_line' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
|    46 |  while ((e = ERR_get_error_line(&file, &line))) {
|       |  ^~~~~
| In file included from /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c:23:
| /home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/include/openssl/err.h:423:15: note: declared here
|   423 | unsigned long ERR_get_error_line(const char **file, int *line);
|       |               ^~~~~~~~~~~~~~~~~~
| /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c: In function 'drain_openssl_errors':
| /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c:60:2: warning: 'ERR_get_error_line' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
|    60 |  while (ERR_get_error_line(&file, &line)) {}

 In file included from /home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/scripts/extract-cert.c:23:
| /home/im/Desktop/yocto_BSB/build/tmp/work/rk3588s-poky-linux/linux-rockchip/5.10/recipe-sysroot-native/usr/include/openssl/err.h:423:15: note: declared here
|   423 | unsigned long ERR_get_error_line(const char **file, int *line);
|       |               ^~~~~~~~~~~~~~~~~~
[... cut ...]
| make[1]: *** [arch/arm64/Makefile:216: rk3588s-orangepi-5.img] Error 2
| make: *** [/home/im/Desktop/yocto_BSB/build/tmp/work-shared/rk3588s/kernel-source/Makefile:185: __sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/im/Desktop/yocto_BSB/build/../meta-rockchip/recipes-kernel/linux/linux-rockchip_5.10.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2956 tasks of which 2568 didn't need to be rerun and 1 failed.

JeffyCN, 2024-08-21:
the dts is not available, please contact your board provider to get the kernel source.

and maybe you can build the kernel manually(not in the yocto), and flash it(boot.img) to the device.

JeffyCN, 2024-08-21:
sorry, i don't know this board, please contact your board's provider.

closed 2024-10-22 by JeffyCN