JeffyCN/meta-rockchip #122: Got SEGV in wayland client.

watatuki, 2024-08-13:
I uses HEAD of scarthgap branch of poky and meta-rockchip.  It run on RK3588 board.
Build core-image-weston with weston-example and gdb.

After booting, weston disktop shell was show.
I run weston weston-simple-egl. As a result, it crash by SIGSEGV.

The gdb result as a shown.

Thread 1 "weston-simple-e" received signal SIGSEGV, Segmentation fault.
wl_list_insert (list=0x10, elm=elm@entry=0x5555605710) at /usr/src/debug/wayland/1.22.0/src/wayland-util.c:47
warning: 47     /usr/src/debug/wayland/1.22.0/src/wayland-util.c: No such file or directory
(gdb) bt
#0  wl_list_insert (list=0x10, elm=elm@entry=0x5555605710) at /usr/src/debug/wayland/1.22.0/src/wayland-util.c:47
#1  0x0000007ff7ee7fc4 in wl_proxy_create_wrapper (proxy=0x5555649970) at /usr/src/debug/wayland/1.22.0/src/wayland-client.c:2456
#2  0x0000007ff1a4c914 in ?? () from /usr/lib/libmali.so.1
#3  0x0000000000000001 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

watatuki, 2024-08-13:
The weston-simple-shm was not crashed.

And "export XDG_RUNTIME_DIR=/run/user/1000" is required by these processes.

JeffyCN, 2024-08-13:
check:
https://github.com/JeffyCN/meta-rockchip/issues/120

watatuki, 2024-08-13:
I apply this patch https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/wayland/0003-client-Assign-proxy-to-default-queue-when-releasing-.patch
As a result, weston-simple-egl do not crash.

Thanks.

closed 2024-08-13 by watatuki