JeffyCN/meta-rockchip #125: recently updated u-boot-rockchip.bb problem

jayshonzs, 2024-08-26:
after adding :
ln -rsf ${UNPACKDIR}/rkbin ${UNPACKDIR}/../

there comes an error:
| ln: failed to create symbolic link '/../rkbin': Permission denied

jayshonzs, 2024-08-26:
And without the new line "ln -rsf ${UNPACKDIR}/rkbin ${UNPACKDIR}/../", there is no error.

JeffyCN, 2024-08-26:
please try other branch, this new var might not be avaliable in older core versions

jayshonzs, 2024-08-26:
> please try other branch, this new var might not be avaliable in older core versions

谢谢，我另外问一个问题

目前rk356x是不是还是推荐使用4.19内核？

因为scarthgap已经不支持4.19的内核了（因为glibc 2.39最老支持5.10内核），所以rk356x是不是还是用kirkstone比较好？

rk356x使用5.10内核是不是有例如libmali驱动上的潜在风险？

麻烦帮我回答一下，谢谢

jayshonzs, 2024-08-26:
> please try other branch, this new var might not be avaliable in older core versions

因为我在rk3566上运行用scarthgap和5.10内核创建的系统，然后在wayland之上，调用glfwCreateWindow()

会出现段错误：

warning: queue 0x55773e84f0 destroyed while proxies still attached:
wl_callback@22 still attached
mali_buffer_sharing@21 still attached
wl_callback@20 still attached
Main process exited, code=dumped, status=11/SEGV
Failed with result 'core-dump'.

jayshonzs, 2024-08-26:
> check:
> https://github.com/JeffyCN/meta-rockchip/commit/3a45b28909e04da7323fb762b6c2866ae7a89ac6
> 
> and:
> https://github.com/JeffyCN/meta-rockchip/issues/120
> (for old kernel and old libmali bug)
> 
> 
> it would be better if you can use the rockchip's newest 6.1 kernel with newer libmali (g13p0)

Thank you, I will try it

closed 2024-10-22 by JeffyCN