JeffyCN/meta-rockchip #128: rk3399 fails to configure glibc on scarthgap

nandra, 2024-10-14:
Hello,

I'm facing an issue when building rk3399pro-evk that glib fails with:

```
| running configure fragment for sysdeps/unix/sysv/linux
| checking installed Linux kernel header files... 3.2.0 or later
| checking for kernel header at least 5.10... too old!
| configure: error: *** The available kernel headers are older than the requested
| *** compatible kernel version
| NOTE: The following config.log files may provide further information.
| NOTE: /data/projects/kas/build/tmp/work/armv8a-poky-linux/glibc/2.39+git/build-aarch64-poky-linux/config.log
| ERROR: configure failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/data/projects/kas/build/../sources/poky/meta/recipes-core/glibc/glibc_2.39
```

JeffyCN, 2024-10-22:
should be fixed in:
https://github.com/JeffyCN/meta-rockchip/commit/1d9f14127cb6ebfe8784d413cb3f53e21e8a5bc0

nandra, 2024-10-22:
I'm on latest origin/scarthgap and still facing this issue

JeffyCN, 2024-10-22:
the oldest header version(5.10) is defined here, the same as libc's kernel header version:
meta-rockchip/conf/machine/include/rockchip-common.inc
28:LINUXLIBCVERSION ?= "5.10-custom%"
30:OLDEST_KERNEL := "${@d.getVar('LINUXLIBCVERSION').split('-')[0]}"

poky/meta/conf/distro/include/tcmode-default.inc
52:PREFERRED_VERSION_linux-libc-headers ?= "${LINUXLIBCVERSION}"

nandra, 2024-10-22:
In rk3399.inc it's overriden to:
```
PREFERRED_VERSION_linux-rockchip := "4.19%"
LINUXLIBCVERSION := "4.19-custom%"
```

nandra, 2024-10-22:
With this change kernel fail to build :(

JeffyCN, 2024-10-22:
well, i thought it would use the final value to calculate oldest_kernel :(

nandra, 2024-10-22:
Can we reopen pls?

JeffyCN, 2024-10-22:
should be fixed now :)

nandra, 2024-10-22:
Thanks but kernel cannot be compiled then:
```
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2700.27-2703.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-bus1: Reference to non-existent node or label "pcfg_pull_up_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1177.26-1180.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2705.27-2711.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-bus4: Reference to non-existent node or label "pcfg_pull_up_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1182.26-1188.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2705.27-2711.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-bus4: Reference to non-existent node or label "pcfg_pull_up_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1182.26-1188.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2705.27-2711.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-bus4: Reference to non-existent node or label "pcfg_pull_up_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1182.26-1188.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2705.27-2711.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-bus4: Reference to non-existent node or label "pcfg_pull_up_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1182.26-1188.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2713.25-2716.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-clk: Reference to non-existent node or label "pcfg_pull_none_10ma"
| 
|   also defined at /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399pro-evb-v11-linux.dts:1190.24-1193.5
| /build/tmp-scarthgap/work-shared/rockchip-rk3399pro-evb/kernel-source/arch/arm64/boot/dts/rockchip/rk3399.dtsi:2718.25-2721.6: ERROR (phandle_references): /pinctrl/sdmmc/sdmmc-cmd: Reference to non-existent node or label "pcfg_pull_up_10ma"
```

JeffyCN, 2024-10-22:
there's something wrong in your kernel dts

nandra, 2024-10-22:
It's stock without any change

JeffyCN, 2024-10-22:
couldn't reproduce it, works well on my side.

JeffyCN, 2025-02-08:
the default kernel of rk3399 is 6.1 now.

closed 2025-02-08 by JeffyCN