JeffyCN/meta-rockchip #132: Unable to build the linux-rochchip 6.1

xbstk, 2024-11-26:
I'd like to add support for new machine.
Firstly, I added file to support the RV1106 processor:
```
require conf/machine/include/arm/armv7a/tune-cortexa7.inc
require conf/machine/include/rockchip-common.inc
require conf/machine/include/mali.inc

RK_SOC_FAMILY = "rv1106g3"

PREFERRED_VERSION_linux-rockchip = "6.1%"
LINUXLIBCVERSION = "6.1-custom%"
```
Then I added file for my machine:
```
require include/rv1106g3.inc

UBOOT_MACHINE = "rv1106_defconfig"

KBUILD_DEFCONFIG = "rv1106_defconfig"
KERNEL_DEVICETREE = "rv1106g3-luckfox-pico-max.dtb"
```
Then, I succedd to build the u-boot:
`MACHINE="luckfox-pico-max" bitbake virtual/bootloader`
but when I'm trying to build the kernel:
`MACHINE="luckfox-pico-max" bitbake virtual/kernel`
I'm encountering the issue:
```
| In file included from /home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/drivers/soc/rockchip/fiq_debugger/fiq_debugger.c:165:
| /home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/drivers/soc/rockchip/fiq_debugger/fiq_debugger_arm.c: In function 'fiq_debugger_dump_stacktrace':
| /home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/drivers/soc/rockchip/fiq_debugger/fiq_debugger_arm.c:241:48: error: implicit declaration of function 'THREAD_INFO' [-Wimplicit-function-declaration]
|   241 |         struct thread_info *real_thread_info = THREAD_INFO(ssp);
|       |                                                ^~~~~~~~~~~
| /home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/drivers/soc/rockchip/fiq_debugger/fiq_debugger_arm.c:241:48: error: initialization of 'struct thread_info *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
| make[6]: *** [/home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/scripts/Makefile.build:250: drivers/soc/rockchip/fiq_debugger/fiq_debugger.o] Error 1
| make[5]: *** [/home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/scripts/Makefile.build:500: drivers/soc/rockchip/fiq_debugger] Error 2
| make[4]: *** [/home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/scripts/Makefile.build:500: drivers/soc/rockchip] Error 2
| make[3]: *** [/home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/scripts/Makefile.build:500: drivers/soc] Error 2
| make[2]: *** [/home/bartiszosti/Documents/xbstk/build-luckfox-pico/tmp/work-shared/luckfox-pico/kernel-source/scripts/Makefile.build:500: drivers] Error 2
| make[2]: *** Waiting for unfinished jobs....
```
Do you know what is wrong?

xbstk, 2024-11-26:
I added a patch to linux-rockchip, which you can find in my fork.
Currently, I have another issue:

�RKUART�DDR V1.10 ec2fae0c96 wesley.yao 22/11/15-10:58:09
S5P1
f967
rgef1
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

U-Boot SPL board init
U-Boot SPL 2017.09 (Jul 10 2024 - 14:42:27)
unrecognized JEDEC id bytes: ff, ef, aa
Trying to boot from MMC2
ENVF: !bad CRC @ 0x200
ENVF: !bad CRC @ 0x200
spl: partition error
Trying to boot from MTD1
ENVF: Primary 0x00000000 - 0x00040000
ENVF: OK
Trying fit image at 0x400 sector
## Verified-boot: 0
## Checking uboot 0x00200000 (lzma @0x00400000) ... sha256(8f1e014ce2...) + sha256(5352c83233...) + OK
## Checking fdt 0x0025bed0 ... sha256(51bba657c3...) + OK
Total: 457.63 ms

What can be wrong with SD card?

JeffyCN, 2024-11-27:
sorry, i don't know much about rv1106.

maybe you can try to follow the readme to use a workable kernel and loader(from the board's provider).

and for the patch status check, you can try to ignore it:
+ERROR_QA:remove = "patch-status

closed 2025-02-08 by JeffyCN