JeffyCN/meta-rockchip #137: Got build error at systemd build in Scarthgap 5.0.6

watatuki, 2025-01-24:
A scarthgap branch of meta-rockchip got build error at systemd build in Scarthgap 5.0.6.

After this patch for systemd, that issue was fixed.  It cause patching failer.
https://git.yoctoproject.org/poky/commit/meta/recipes-core/systemd?h=scarthgap&id=fb691445b2ff00138974206342015e8f1741b9fe

0001-meson-do-not-fail-build-with-newer-kernel-headers.patch is not nessesally for latest scarthgap.

**Error message**
ERROR: systemd-1_255.13-r0 do_patch: Applying patch '0001-meson-do-not-fail-build-with-newer-kernel-headers.patch' on target directory '/build/tmp/work/aarch64-agl-linux/systemd/255.13/git'
CmdError('quilt --quiltrc /build/tmp/work/aarch64-agl-linux/systemd/255.13/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-meson-do-not-fail-build-with-newer-kernel-headers.patch
patching file src/basic/meson.build
Hunk #1 FAILED at 240.
1 out of 1 hunk FAILED -- rejects in file src/basic/meson.build
Patch 0001-meson-do-not-fail-build-with-newer-kernel-headers.patch can be reverse-applied

stderr: ')
ERROR: Logfile of failure stored in: /build/tmp/work/aarch64-agl-linux/systemd/255.13/temp/log.do_patch.696662
ERROR: Task (/poky/meta/recipes-core/systemd/systemd_255.13.bb:do_patch) failed with exit code '1'

JeffyCN, 2025-01-25:
ok, i've uploaded a patch for that, thanx.

watatuki, 2025-01-25:
Thanks.

I think so, SRC_URI:append need to remove from systemd_%.bbappend.

JeffyCN, 2025-01-25:
oops, done :)

closed 2025-01-26 by watatuki