JeffyCN/meta-rockchip #138: Secure ADB

DhimanAirao, 2025-02-07:
Hii, I am using yocto build for my rk3568 dev board and I am using ADB. Currently when i do "adb" it directly enters as root user. I want to secure this so that any third party cannot access adb and use my device as root. how to do it?

JeffyCN, 2025-02-08:
The official Linux version of adbd doesn't support it, but I have some patches in Rockchip's Buildroot SDK for password or RSA key verification.

This is a prebuilt adbd used in Rockchip's Yocto SDK:
https://filebin.net/qrh7pz7qztainqwb/yocto_adbd.tgz

You can try to remove the original adbd in Yocto:
find /usr -o -name "android-tools-adbd*" -print0 \
        -o -name "android-gadget*" -print0 \
        -o -name "adbd.sh" -print0 2>/dev/null | xargs -0 rm -rf
And replace it with this version.

JeffyCN, 2025-02-10:
that tarball is for sysvinit not systemd.

the prebuilt adbd would check ADB_SECURE environment to enable security, so you may need to set ADB_SECURE=1 before running adbd.

or you can use this one(hard-coded to enable security):
https://filebin.net/ofti5doj20zxm9kh/adbd

DhimanAirao, 2025-02-10:
Hii, Thanks for helping me with this. It worked with tar folder you shared. I just needed to export ADB variable in init.d service file. It will be great if you integrate it in meta-rockchip.

JeffyCN, 2025-02-11:
you can modify the adbd-auth.sh, to login/switch to that user and start a sub-shell, and exit 1 after that.

for example:
change the 'exit 0' to '/bin/login; exit 1'

DhimanAirao, 2025-02-12:
I added this now it always ask me for username and password during "adb shell" but now I am not able to use "adb push" it throws me error

[image]

JeffyCN, 2025-02-12:
the other way is to set env ADBD_SHELL to a custom script, and switch user in it

DhimanAirao, 2025-02-12:
`#!/bin/bash

# Run /bin/login to authenticate the user
/bin/login

# If authentication succeeds, set ADBD_SHELL to /bin/bash
export ADBD_SHELL="/bin/bash"

# Start an interactive shell
exec /bin/bash
`
Something like this??
 And run this script before starting ADB service??

JeffyCN, 2025-02-12:
no, just something like:

export ADBD_SHELL=/bin/login
before running adbd.

or ADBD_SHELL=/usr/bin/adbd-shell.sh

adbd-shell.sh:
#!/bin/sh
# switch user
/bin/bash

closed 2025-03-12 by JeffyCN