JeffyCN/meta-rockchip #141: rockchip-rkaiq install fails when enabling multilib

vbpieter, 2025-03-11:
When i enable multilib to build 32-bit libraries alongside 64-bit, the rkaiq recipe fails to install

```bb
# conf/local.conf
require conf/multilib.conf
MULTILIBS = "multilib:lib32"
DEFAULTTUNE:virtclass-multilib-lib32 = "armv7ahf"
```
the output: [log.do_install.txt](https://github.com/user-attachments/files/19186254/log.do_install.txt)

I've added some print statements and it seems the issue is in the `chrpath -d ${D}/${libdir}/libsmartIr.so` line:
```bash
open: No such file or directory
elf_open: Invalid argument
```

```bb
# recipes-multimedia/rockchip-rkaiq/rockchip-rkaiq.bb
do_install:append () {
    bbwarn "DO INSTALL APPEND REACHED"
	# rkaiq installed 3A server to the wrong dir.
	[ ! -d ${D}/usr/usr ] || cp -rp ${D}/usr/usr ${D}/
    bbwarn "DO INSTALL APPEND REACHED 2"

	# Drop unused tools
	rm -rf ${D}/usr/etc ${D}/usr/usr ${D}/usr/bin/*demo \
		${D}/usr/bin/rkaiq_tool_server ${D}/usr/bin/dumpcam

    bbwarn "DO INSTALL APPEND REACHED 3"
	chrpath -d ${D}/${libdir}/libsmartIr.so

    bbwarn "DO INSTALL APPEND REACHED 4"
	install -d ${D}${sysconfdir}/iqfiles
	ln -sf isp3x ${S}/rkaiq/iqfiles/isp30
    bbwarn "DO INSTALL APPEND REACHED 5"

	IQFILES_DIR="$(echo isp${RK_ISP_VERSION} | tr 'A-Z' 'a-z' | tr -d '.')"
	install -m 0644 ${S}/rkaiq/iqfiles/$IQFILES_DIR/*.json \
		${D}${sysconfdir}/iqfiles/

    bbwarn "DO INSTALL APPEND REACHED 6"
	install -d ${D}${sysconfdir}/init.d
	install -m 0755 ${WORKDIR}/rkaiq_daemons.sh ${D}${sysconfdir}/init.d/
    bbwarn "DO INSTALL APPEND REACHED 7"
}
```

Any idea how to to properly fix this?

JeffyCN, 2025-03-12:
maybe try:
```

+++ b/recipes-multimedia/rockchip-rkaiq/rockchip-rkaiq.bb
@@ -57,7 +57,7 @@ do_install:append () {
        rm -rf ${D}/usr/etc ${D}/usr/usr ${D}/usr/bin/*demo \
                ${D}/usr/bin/rkaiq_tool_server ${D}/usr/bin/dumpcam

-       chrpath -d ${D}/${libdir}/libsmartIr.so
+       chrpath -d ${D}/*/libsmartIr.so

```

JeffyCN, 2025-03-12:
and please provide the run.do_install.txt as well.

vbpieter, 2025-03-12:
Your recommendation made me look into the `${libdir}` value, it resolves to `/usr/lib64` instead of the expected `/usr/lib` (most likely due to enabling multilib since I am building both 32 and 64 bit libraries for some recipes).

The `chrpath` commando fails since GNU Make installs everything in `/usr/lib` instead of the now 'expected' `/usr/lib64`

This fix seems to work (camera looks good) with minimal changes:
```diff
--- a/recipes-multimedia/rockchip-rkaiq/rockchip-rkaiq.bb
+++ b/recipes-multimedia/rockchip-rkaiq/rockchip-rkaiq.bb
@@ -57,7 +57,7 @@ do_install:append () {
        rm -rf ${D}/usr/etc ${D}/usr/usr ${D}/usr/bin/*demo \
                ${D}/usr/bin/rkaiq_tool_server ${D}/usr/bin/dumpcam
 
-       chrpath -d ${D}/${libdir}/libsmartIr.so
+    chrpath -d ${D}/usr/lib/libsmartIr.so
 
        install -d ${D}${sysconfdir}/iqfiles
        ln -sf isp3x ${S}/rkaiq/iqfiles/isp30
@@ -68,6 +68,16 @@ do_install:append () {
 
        install -d ${D}${sysconfdir}/init.d
        install -m 0755 ${WORKDIR}/rkaiq_daemons.sh ${D}${sysconfdir}/init.d/
+
+    # Move static and shared libraries to the correct libdir
+    if [ -d "${D}/usr/lib" ]; then
+        mkdir -p ${D}${libdir}
+        mv ${D}/usr/lib/*.a ${D}${libdir}/ || true
+        mv ${D}/usr/lib/*.so ${D}${libdir}/ || true
+
+        # Remove the /usr/lib directory (if empty) to prevent packaging issues
+        rmdir --ignore-fail-on-non-empty ${D}/usr/lib
+    fi
 }
```
All files need to be moved to `/usr/lib64` since you get a QA Issue otherwise:
```bash
ERROR: rockchip-rkaiq-1.0-r0 do_package: QA Issue: rockchip-rkaiq: Files/directories were installed but not shipped in any package:
  /usr/lib/libIspFec.a
  /usr/lib/librkaiq.a
  /usr/lib/libsmartIr.a
  /usr/lib/libIspFec.so
  /usr/lib/libsmartIr.so
  /usr/lib/librkaiq.so
  /usr/lib/librkrawstream.so
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
rockchip-rkaiq: 7 installed and not shipped files. [installed-vs-shipped]
ERROR: rockchip-rkaiq-1.0-r0 do_package: Fatal QA errors were found, failing task.
```
And after adding those files to the `FILES:${PN}` variable:
```bash
ERROR: rockchip-rkaiq-1.0-r0 do_package_qa: QA Issue: non -staticdev package contains static .a library: rockchip-rkaiq path '/usr/lib/libIspFec.a'
non -staticdev package contains static .a library: rockchip-rkaiq path '/usr/lib/librkaiq.a'
non -staticdev package contains static .a library: rockchip-rkaiq path '/usr/lib/libsmartIr.a' [staticdev]
WARNING: rockchip-rkaiq-1.0-r0 do_package_qa: QA Issue: rockchip-rkaiq-dbg: found library in wrong location: /usr/lib/.debug/librkaiq.so
rockchip-rkaiq-dbg: found library in wrong location: /usr/lib/.debug/librkrawstream.so
rockchip-rkaiq: found library in wrong location: /usr/lib/libIspFec.so
rockchip-rkaiq: found library in wrong location: /usr/lib/libsmartIr.so
rockchip-rkaiq: found library in wrong location: /usr/lib/librkaiq.so
rockchip-rkaiq: found library in wrong location: /usr/lib/librkrawstream.so [libdir]
ERROR: rockchip-rkaiq-1.0-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /media/ssda/dev/yocto/build-tbs/tmp/work/rk3588s_tbs-poky-linux/rockchip-rkaiq/1.0/temp/log.do_package_qa.56263
ERROR: Task (/media/ssda/dev/yocto/meta-rockchip/recipes-multimedia/rockchip-rkaiq/rockchip-rkaiq.bb:do_package_qa) failed with exit code '1'
```

Thanks a lot for the fast reply and pointing me into the right direction!

JeffyCN, 2025-03-12:
Thanks for fixing it.

So likely something wrong in the rkaiq cmake script, maybe it hard-coded the dest dir to '/usr/lib'.

Can you submit a pull request?

Maybe we can try to do something like:
```
do_install:append () {
        if [ "${libdir}" != "/usr/lib" ]; then
               // move those libs
        fi

        rm -rf ${D}/usr/etc ${D}/usr/usr ${D}/usr/bin/*demo \
                ${D}/usr/bin/rkaiq_tool_server ${D}/usr/bin/dumpcam
 
        chrpath -d ${D}/${libdir}/libsmartIr.so
       ...
```

vbpieter, 2025-03-12:
Yeah it seems like it is hardcoded in the rkaiq cmake script, after quickly scouting it:
```cmake
# rkaiq_3A_server/CMakeLists.txt
if(BUILDROOT_BUILD_PROJECT)
	install(TARGETS rkaiq_3A_server
		RUNTIME DESTINATION /usr/bin
		LIBRARY DESTINATION /usr/lib
		)

	install(FILES S40rkaiq_3A DESTINATION /etc/init.d/
		PERMISSIONS OWNER_EXECUTE OWNER_WRITE OWNER_READ
		GROUP_EXECUTE GROUP_READ
		WORLD_EXECUTE WORLD_READ)
else()
	install(TARGETS rkaiq_3A_server
		RUNTIME DESTINATION "usr/bin"
		LIBRARY DESTINATION "usr/lib"
		)

	install(FILES S40rkaiq_3A DESTINATION "etc/init.d"
		PERMISSIONS OWNER_EXECUTE OWNER_WRITE OWNER_READ
		GROUP_EXECUTE GROUP_READ
		WORLD_EXECUTE WORLD_READ)
endif()
```
I will submit a pull request

closed 2025-03-12 by JeffyCN