JeffyCN/meta-rockchip #143: RK_UBOOT_SPL variable in u-boot-rockchip not set

ThomasVandenabeele, 2025-04-10:
I'm building my yocto for the RK3506B chip. However, I think this issue could be addressing multiple chipsets.

When using the meta-rockchip as described in the README, I noticed the u-boot-rockchip.bb recipe (recipes-bsp/u-boot/u-boot-rockchip.bb) relies on  a variable $RK_UBOOT_SPL on line 76.
The variable is set in the machine configuration conf/machine/rockchip-rk3506-evb.conf. However, not set when yocto compiles the u-boot-rockchip recipe.

I verified this by adding bbwarn statements inside the if-else statement and tried printing the $RK_UBOOT_SPL variable just before the if-statement.
The variable bbwarn shows nothing and the statement always go inside the else-statement using the Rockchip Miniloader instead of the U-Boot SPL.

When I add the following to the local.conf file, the variable is set and use correctly the U-Boot SPL in u-boot-rockchip.bb recipe:
`export RK_UBOOT_SPL = "1"`

Is it possible that I'm doing something wrong?

JeffyCN, 2025-04-11:
thanks for noticing that, should be fixed now.

ThomasVandenabeele, 2025-04-11:
Great, now it is using the U-Boot SPL after the changes in https://github.com/JeffyCN/meta-rockchip/commit/bc7b54365898f5973c12a2802809ca497e04f23e !

Thank you.