JeffyCN/meta-rockchip #149: afptool: not found

tickmoo, 2025-07-17:
ERROR: core-image-minimal-1.0-r0 do_image_complete: Execution of '/home/boyebio/yocto/github/build/tmp/work/rockchip_rk3588_evb-poky-linux/core-image-minimal/1.0/temp/run.gen_rkupdateimg.8971' failed with exit code 127
ERROR: Logfile of failure stored in: /home/boyebio/yocto/github/build/tmp/work/rockchip_rk3588_evb-poky-linux/core-image-minimal/1.0/temp/log.do_image_complete.8971
Log data follows:
| DEBUG: Executing python function sstate_task_prefunc
| DEBUG: Python function sstate_task_prefunc finished
| DEBUG: Executing python function do_image_complete
| NOTE: Executing link_rootfs_image ...
| DEBUG: Executing shell function link_rootfs_image
| DEBUG: Shell function link_rootfs_image finished
| NOTE: Executing do_image_wic_ufs ...
| DEBUG: Executing shell function do_image_wic_ufs
| Creating wic image for UFS(4K logical sector size)...
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| GPT data structures destroyed! You may now partition the disk using fdisk or
| other utilities.
| Adding part-1: uboot 512-1535 80E124B8-F98A-4E9A-A112-4A19D2B25326...
| Creating new GPT entries in memory.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Adding part-2: boot 1536-5791 FA722F62-10EF-46A1-89C5-947C004CD54F...
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Adding part-3: rootfs 6144-14530 614E0000-0000-4B53-8000-1D28000054A9...
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| Warning: The kernel is still using the old partition table.
| The new table will be used at the next reboot or after you
| run partprobe(8) or kpartx(8)
| The operation has completed successfully.
| 32+0 records in
| 32+0 records out
| 32768 bytes (33 kB, 32 KiB) copied, 0.000192923 s, 170 MB/s
| 32+0 records in
| 32+0 records out
| 32768 bytes (33 kB, 32 KiB) copied, 9.2011e-05 s, 356 MB/s
| 1+0 records in
| 1+0 records out
| 512 bytes copied, 5.4047e-05 s, 9.5 MB/s
| 1+0 records in
| 1+0 records out
| 512 bytes copied, 0.000146302 s, 3.5 MB/s
| 32+0 records in
| 32+0 records out
| 16384 bytes (16 kB, 16 KiB) copied, 0.000163407 s, 100 MB/s
| 512+0 records in
| 512+0 records out
| 524288 bytes (524 kB, 512 KiB) copied, 0.000633464 s, 828 MB/s
| DEBUG: Shell function do_image_wic_ufs finished
| NOTE: Executing gen_rkparameter ...
| DEBUG: Executing shell function gen_rkparameter
| Generating core-image-minimal-rockchip-rk3588-evb.rootfs.parameter...
| DEBUG: Shell function gen_rkparameter finished
| NOTE: Executing gen_rkupdateimg ...
| DEBUG: Executing shell function gen_rkupdateimg
| Generating core-image-minimal-rockchip-rk3588-evb.rootfs.package-file...
| WARNING: exit code 127 from a shell command.
| /home/boyebio/yocto/github/build/tmp/work/rockchip_rk3588_evb-poky-linux/core-image-minimal/1.0/temp/run.gen_rkupdateimg.8971: 197: afptool: not found
| DEBUG: Python function do_image_complete finished
ERROR: Task (/home/boyebio/yocto/github/build/../poky/meta/recipes-core/images/core-image-minimal.bb:do_image_complete) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3741 tasks of which 3720 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/boyebio/yocto/github/build/../poky/meta/recipes-core/images/core-image-minimal.bb:do_image_complete
    log: /home/boyebio/yocto/github/build/tmp/work/rockchip_rk3588_evb-poky-linux/core-image-minimal/1.0/temp/log.do_image_complete.8971
Summary: There was 1 ERROR message, returning a non-zero exit code.

Sapieron, 2025-07-18:
Checked out `meta-rockchip` to newest master `ab0273aff95731deb1ae1bd9a52ac87b45722095`, same issue as OP.

JeffyCN, 2025-07-18:
Which version of Poky did you test? (I was testing Scarthgap.) Could you please upload  build/tmp/work/x86_64-linux/rk-binary-native/1.0/temp ? I can check the logs for more details.

Nettoysfkf, 2025-07-19:
According to the ${bindir} definition in [bitbake.conf](https://github.com/yoctoproject/poky/blob/master/meta/conf/bitbake.conf#L45), maybe your some native change affect on the ${exec_prefix}.

JeffyCN, 2025-08-05:
Should be fixed now, thanks for checking it!

closed 2025-08-05 by JeffyCN