JeffyCN/meta-rockchip #152: fcntlcmds issue with rockchip-rk3576-evb build

mb2012-boop, 2025-08-06:
Hi,

I am trying to build a minimal image (core-image-minimal) for the rockchip-rk3576-evb. Here are my configurations (on the kirkstone branch):

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "rockchip-rk3576-evb"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.28"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa72-cortexa53"
TARGET_FPU           = ""
meta-rockchip        = "kirkstone:3fc8aa9f366a26a27edd96cfde4bce15add36834"
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:f29a38bdcb908a75b8314b500802ffa651f976ed"
meta-oe              = "kirkstone:06fc0278f10d630838d703dde707bbf0e2999873"

Every time I try to build the image the build fails with an error at around 60% of the build relating to strace:

cho '../../strace-5.16/src/'`fetch_indirect_syscall_args.c
| In file included from ../../strace-5.16/src/defs.h:404,
|                  from ../../strace-5.16/src/fcntl.c:12:
| ../../strace-5.16/src/xlat/fcntlcmds.h:54:7: error: 'F_GETLK64' undeclared here (not in a function); did you mean 'F_GETLK'?
|    54 |  XLAT(F_GETLK64),
|       |       ^~~~~~~~~
| ../../strace-5.16/src/xlat.h:64:54: note: in definition of macro 'XLAT'
|    64 | # define XLAT(val)                      { (unsigned)(val), #val }
|       |                                                      ^~~
| ../../strace-5.16/src/xlat/fcntlcmds.h:57:7: error: 'F_SETLK64' undeclared here (not in a function); did you mean 'F_SETLK'?
|    57 |  XLAT(F_SETLK64),
|       |       ^~~~~~~~~
| ../../strace-5.16/src/xlat.h:64:54: note: in definition of macro 'XLAT'
|    64 | # define XLAT(val)                      { (unsigned)(val), #val }
|       |                                                      ^~~
| ../../strace-5.16/src/xlat/fcntlcmds.h:60:7: error: 'F_SETLKW64' undeclared here (not in a function); did you mean 'F_SETLKW'?
|    60 |  XLAT(F_SETLKW64),
|       |       ^~~~~~~~~~
| ../../strace-5.16/src/xlat.h:64:54: note: in definition of macro 'XLAT'
|    64 | # define XLAT(val)                      { (unsigned)(val), #val }
|       |                                                      ^~~
| make[3]: *** [Makefile:4958: libstrace_a-fcntl.o] Error 1
| make[3]: *** Waiting for unfinished jobs....
| make[3]: Leaving directory '/home/yocto/build/tmp/work/cortexa72-cortexa53-poky-linux/strace/5.16-r0/build/src'
| rm ioctlsort1.o ioctls_all1.h ioctls_all0.h ioctlsort1 ioctlsort0 ioctlsort0.o
| make[2]: Leaving directory '/home/yocto/build/tmp/work/cortexa72-cortexa53-poky-linux/strace/5.16-r0/build/src'
| make[2]: *** [Makefile:2642: all] Error 2
| ERROR: oe_runmake failed
| make[1]: Leaving directory '/home/yocto/build/tmp/work/cortexa72-cortexa53-poky-linux/strace/5.16-r0/build'
| WARNING: exit code 1 from a shell command.
| make[1]: *** [Makefile:595: all-recursive] Error 1
| make: *** [Makefile:500: all] Error 2

JeffyCN, 2025-08-06:
Checking strace's changelog, it looks like to be fixed in newer versions.

So we might need to downgrade libc header or upgrade strace or backport the strace patch.

Maybe try to switch to 5.10-custom?

mb2012-boop, 2025-08-07:
Thank you. This seems to have resolved the strace error, but another error then happens further in the build, I think it's related to the rk3576_defconfig.

/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -Wp,-MD,scripts/kconfig/.zconf.tab.o.d -Iscripts/kconfig -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -DCURSES_LOC="<curses.h>" -DLOCALE  -I/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig -Iscripts/kconfig -c -o scripts/kconfig/zconf.tab.o scripts/kconfig/zconf.tab.c
| In file included from scripts/kconfig/zconf.tab.c:2468:
| /home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig/confdata.c: In function 'conf_write':
| /home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig/confdata.c:771:19: warning: '%s' directive writing likely 7 or more bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
|   771 |  sprintf(newname, "%s%s", dirname, basename);
|       |                   ^~~~~~
| /home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig/confdata.c:771:19: note: assuming directive output of 7 bytes
| In file included from /usr/include/stdio.h:867,
|                  from scripts/kconfig/zconf.tab.c:82:
| /usr/include/x86_64-linux-gnu/bits/stdio2.h:36:10: note: '__builtin___sprintf_chk' output 1 or more bytes (assuming 4104) into a destination of size 4097
|    36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|    37 |       __bos (__s), __fmt, __va_arg_pack ());
|       |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| In file included from scripts/kconfig/zconf.tab.c:2468:
| /home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig/confdata.c:774:20: warning: '.tmpconfig.' directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
|   774 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
|       |                    ^~~~~~~~~~~~~~~~~
| In file included from /usr/include/stdio.h:867,
|                  from scripts/kconfig/zconf.tab.c:82:
| /usr/include/x86_64-linux-gnu/bits/stdio2.h:36:10: note: '__builtin___sprintf_chk' output between 13 and 4119 bytes into a destination of size 4097
|    36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|    37 |       __bos (__s), __fmt, __va_arg_pack ());
|       |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
|   gcc  -isystem/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/usr/lib                         -L/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/yocto/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2  -o scripts/kconfig/conf scripts/kconfig/conf.o scripts/kconfig/zconf.tab.o
| scripts/kconfig/conf  --defconfig=arch/../configs/rk3576_defconfig Kconfig
| ***
| *** Can't find default configuration "arch/../configs/rk3576_defconfig"!
| ***
| make[2]: *** [/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/kconfig/Makefile:121: rk3576_defconfig] Error 1
| make[1]: Leaving directory '/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/build'
| make[1]: *** [/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git/Makefile:486: rk3576_defconfig] Error 2
| make: Leaving directory '/home/yocto/build/tmp/work/rockchip_rk3576_evb-poky-linux/u-boot-rockchip/1_2017.09-r0/git'
| make: *** [Makefile:157: sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.

mb2012-boop, 2025-08-08:
Thank you for the help, I will close this now as it looks like the strace issue is resolved.

closed 2025-08-08 by mb2012-boop